rust-embedded/rust-i2cdev #3: Crate 0.1 SMBus functionality not exposed

posborne, 2015-09-12:
Just creating a note as I discovered this while adding some additional example code.  A new version should be released soon that fixes this.

rundrop1, 2015-09-13:
Thanks!  I hadn't come across it yet, if you'd like in a couple of days I should have another example if you'd like to link it or add it or whoever can just follow this comment to my repository when it's up!  

I learned a lot from your code, thanks!

posborne, 2015-09-14:
I'm also in the process of adding some additional examples to the library to continue to see where the pain points are, starting with the MPL115A2 pressure/tempeature sensor.  Based on this work, I may end up adding an abstraction on top of registers (maybe a macro)  that should allow users to use a higher level abstraction when supporting devices.  The low-level interface will always also be available as devices are all over the map as far as what is required to talk to them.

posborne, 2015-09-21:
Fixed in master, so closing.  I am in the process of changing around quite a few things that should help the lib to support things beyond linux (e.g. microcontrollers, unit tests mocks).  Eventually, I would like to add support for zinc.  This may involve splitting the library into separate crates with the core being no_std.

closed 2015-09-21 by posborne