rust-embedded/rust-i2cdev #9: Examples in README are not correct any more

posborne, 2015-10-07:


rundrop1, 2015-10-09:
Hello, are you accepting PR's for examples?  Or even additions to /sensors?  I know you are planning on incorporating zinc/zinc like structure are you planning on deprecating LinuxI2CDevice?

posborne, 2015-10-09:
Hello @rundrop1.  PRs for both examples and new sensors would be great!

There are no plans to deprecate LinuxI2CDevice.  With the recent changes, it is now possible for sensors to be written that only depend on the I2CDevice trait and not on the linux one directly.  As long as you follow the style of the other sensors, things should work out well.  The new design also allows for unit testing.

rundrop1, 2015-10-09:
Awesome! Very stoked to push a request, am trying to get it up tonight!

posborne, 2016-03-16:
Hi @rundrop1, I know it has been a very long time.  Did you ever end up putting together any changes for this or other parts of the library?  Did the crate work for your use cases?

rundrop1, 2016-03-16:
The lib works wonderfully! I didn't need to make any changes.  Very excited to see Rust moving to more embedded stuffs!  At some point I'll clean up my PCA9685 driver but I'm in the middle of something else at the moment!

Thanks for all the work you've done on the lib.

closed 2016-03-16 by posborne