rust-embedded/rust-i2cdev #17: error: Could not compile `nix`.

Goyapa, 2016-06-03:
Hello,
thanks for this "rust-i2cdev" crate, happy to try it out,
but i have an issue.

Short version:
please let update to "nix 0.5"

Long version:
i've tried to compile with this cargo.toml on a RaspberryPi 3

```
[dependencies]
i2cdev = "0.2.0"

[dev-dependencies]
docopt = "0.6.80"

```

```
pi@raspberrypi:~/rustlang-sample/wiinunchuck-test $ cargo run
    Updating registry `https://github.com/rust-lang/crates.io-index`
   Compiling nix v0.4.2
/home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/nix-0.4.2/src/fcntl.rs:102:28: 102:41 error: mismatched types [E0308]
/home/pi/.cargo/registry/src/github.com-1ecc6299db9ec823/nix-0.4.2/src/fcntl.rs:102         unsafe { ffi::open(cstr.as_ptr(),
oflag.bits(), mode.bits() as mode_t) }
...
error: aborting due to 24 previous errors
error: Could not compile `nix`.
```

at 
https://crates.io/crates/i2cdev

i see that the dependence is
nix ^0.4
so i'm not able to choose nix 0.5 but would work fine
which i have tested with my setup in a project.

```
pi@raspberrypi:~/rustlang-sample/wiinunchuck-test $ rustup show
installed toolchains
--------------------

stable-armv7-unknown-linux-gnueabihf
nightly-armv7-unknown-linux-gnueabihf (default)

active toolchain
----------------

nightly-armv7-unknown-linux-gnueabihf (default)
rustc 1.11.0-nightly (915b003e3 2016-06-02)
```

posborne, 2016-06-03:
Ah, we should definitely be able to bump.  Thanks for reporting.

posborne, 2016-06-03:
Actually, it looks like we have but need to release a new version.

Goyapa, 2016-06-03:
what is needed to bump from i2cdev = "0.2.0" to "0.2.1" on crates.io?

posborne, 2016-06-06:
I need to publish a new release, which I will be doing shortly.

In the future, an alternative workaround is to point your crate at Github directly like this:

    [dependencies]
    i2cdev = { git = "https://github.com/rust-embedded/rust-i2cdev.git" }

This is obviously not what you want long term.  New release will be out shortly.

posborne, 2016-06-06:
Version 0.3.0 is now released.  I moved to 0.3.0 as there are some API changes requiring that the major version be bumped according to semver.  It appears that docs are not updating now that I have moved things to the rust-embedded org, so I will see about getting those updated ASAP.

closed 2016-06-06 by posborne