rust-embedded/rust-i2cdev #25: IO traits

Kixunil, 2016-11-10:
You have `Read` trait checked in README checklist but I can't find it. Did you forget to commit something?

posborne, 2016-11-10:
Hmmm, I think I ended up implementing `read`/`write` functions that return `I2CResult` directly on the trait but not implementing `io::Read/Write`.  What do you think the major benefits of supporting the `io` traits would be for I2C specifically?  I think I would consider making a change but motivation is always good to keep in mind.

nastevens, 2016-11-16:
With I2C's packetized nature, I don't know that it really makes sense to implement `Read` and `Write`. The one situation I can see where that would make more sense is for Flash and EEPROMs, but in that case, you're still going to need a higher-level driver to deal with the particular part. In that case, I would expect the higher-level driver to be the one implementing `Read` and `Write`.

I'm not opposed to implementing `Read`/`Write` for this library, but at this point I'm not really sure what those implementations would do!

Kixunil, 2016-11-17:
I gave it second thought and realized, that I2C really isn't stream, so it doesn't make much sense. I wanted some abstraction layer for project I'm making but I might do it other way.

BTW, what's the relation between `read`/`write` functions and `smbus*` functions in I2C crate? Would it make sense to create "I2C only" and "smbus only" traits? They seem like different buses to me even though they are similar.

closed 2018-03-20 by posborne