rust-embedded/rust-i2cdev #26: Use io::Error

Kixunil, 2016-11-10:
I believe it's better to use `io::Error` because it's standard for these kinds of operations. What do you think?

posborne, 2016-11-10:
Although standard, there is information loss which I don't think is a good thing.  We could certainly define a transformation into an io Error for an i2c error, but I'm not convinced it makes sense to start with that information loss.

nastevens, 2016-11-16:
I agree with @posborne - because the I2C driver can interact with more than just a file stream, making `io::Error` the default would potentially mask errors. There is already a `From` impl for `LinuxI2CError`, so converting to an `io::Error` is as simple as a `io::Error::from(<error>)` or `<error>.into()`

I move to close this issue unless there's a different compelling reason to convert.

Kixunil, 2016-11-17:
I prefer not losing error information too. I find `io::Error` somewhat unfortunate decision. I like how `I2CDevice` trait expresses it. However, it might still be desirable to make the conversion. What do you think about adding another bound to guarantee that `Error` can be inspected somehow?

For example some trait (let's say `I2CError`) with `fn error_kind() -> ErrorKind` (It may be `io::ErrorKind` or something custom). Then blanket `impl <T: I2CError> From<T> for io::Error` could be writen too.

Kixunil, 2017-03-27:
Thinking about it again and I believe current state is good. Closing.

closed 2017-03-27 by Kixunil