rust-embedded/rust-i2cdev #31: Unable to read from open bus?

RandomInsano, 2017-01-21:
I'm attempting to read data from a [CHIP](https://getchip.com/pages/chip) computer. There are i2c calls on bus 0 that allow you to read the battery information and querying that information is easily done with [i2cget and i2cset](https://github.com/NextThingCo/CHIP-hwtest/blob/chip/stable/chip-hwtest/bin/battery.sh), and those work great on the bus that's likely opened by some other app.

While trying to do the same with i2cdev, I reliably get EBUSY while trying to set the active device address. I peeled apart some of i2cget's source code, but I can't quite figure out why it is able to query the device while this library can't.

Any thoughts or testing I can do? It seems to attach fine to the idle busses 1 and 2 on the SOC, just not bus 0.

Example code:
```rust
extern crate i2cdev;

use i2cdev::core::*;
use i2cdev::linux::{LinuxI2CDevice, LinuxI2CError};

const ADDRESS : u16 = 0x34;
const BUS: &'static str = "/dev/i2c-0";

fn main() {
      // Dies here:
        let mut dev = LinuxI2CDevice::new(&BUS, 0).unwrap();
        println!("Opened successfully");
        let result = dev.smbus_read_byte_data(0x00).unwrap();
        println!("Response was {}.", result);
}
```

RandomInsano, 2017-01-21:
Ugh. Just noticed that the CHIP script is forcing access to the bus. I'll patch mine for that IOCTL, but it may not be safe to offer consumers of this library.

closed 2017-01-21 by RandomInsano

posborne, 2017-01-21:
@RandomInsano which ioctl is this is it using to force access?  I would like to have good support for CHIP if at all possible (obviously, without forcing hacks that would be detrimental elsewhere).

RandomInsano, 2017-01-22:
It's fairly simple, just use `I2C_SLAVE_FORCE` instead of just `I2C_SLAVE`.

I made a dirty hack here:
https://github.com/RandomInsano/rust-i2cdev/commit/84ec6fae3b29a8361293defe68dcbb18ed0e0860?diff=unified

Now, it's mentioned in the man page for i2cget that forcing is bad, and in my case, I'm interfacing with the IC that controls power. I don't *think* it could be possible for the master to throw out corrupted data, but it'll be worth warning developers. I have five chips, so if one lights on fire I have a few backups :D

```
       -f     Force  access  to  the  device  even  if it is already busy. By default, i2cget will refuse to access a
              device which is already under the control of a kernel driver. Using this  flag  is  dangerous,  it  can
              seriously  confuse  the kernel driver in question. It can also cause i2cget to return an invalid value.
              So use at your own risk and only if you know what you're doing.

```

posborne, 2017-01-23:
Thanks for the explanation.  That doesn't seem to be an issue with other devices on the CHIP or something in the kernel that is not properly sharing the bus.  For now (until more people complain), it doesn't seem like a priority to support that ioctl here.

Let me know if there are other threads that include discussion on why this might be an OK thing to do.

RandomInsano, 2017-01-23:
Yeah, likely a safe bet. I'll maintain the fork for now and hack on things over there.

thegecko, 2017-02-14:
Thanks @RandomInsano, I came across the same problem and needed to force access in this way.

emcfarlane, 2020-12-11:
Thanks @RandomInsano, same issue on coral.ai dev board. Added a new fork here: https://github.com/emcfarlane/rust-i2cdev/commit/bf3800f5da44b35f13c26228ea6b061d08d4e9b2