rust-embedded/rust-i2cdev #34: Problem with missing macro ioctl!

MrSpock, 2017-06-23:
Hi,

Any compilation attempt for crate ends up with following errors:

```
spock@macspock:~/OneDrive/devel/vl53l0x> ck
   Compiling i2cdev v0.3.1
error: cannot find macro `ioctl!` in this scope
   --> /Users/spock/.cargo/registry/src/github.com-1ecc6299db9ec823/i2cdev-0.3.1/src/ffi.rs:175:1
    |
175 | ioctl!(bad ioctl_i2c_smbus with I2C_SMBUS);
    | ^^^^^

error: cannot find macro `ioctl!` in this scope
   --> /Users/spock/.cargo/registry/src/github.com-1ecc6299db9ec823/i2cdev-0.3.1/src/ffi.rs:174:1
    |
174 | ioctl!(bad ioctl_set_i2c_slave_address with I2C_SLAVE);
    | ^^^^^

error[E0425]: cannot find function `ioctl_set_i2c_slave_address` in this scope
   --> /Users/spock/.cargo/registry/src/github.com-1ecc6299db9ec823/i2cdev-0.3.1/src/ffi.rs:182:9
    |
182 |         ioctl_set_i2c_slave_address(fd, slave_address as *mut u8)
    |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^ did you mean `i2c_set_slave_address`?

error[E0425]: cannot find function `ioctl_i2c_smbus` in this scope
   --> /Users/spock/.cargo/registry/src/github.com-1ecc6299db9ec823/i2cdev-0.3.1/src/ffi.rs:202:5
    |
202 |     ioctl_i2c_smbus(fd, p_args).map(drop)
    |     ^^^^^^^^^^^^^^^ not found in this scope

error: aborting due to previous error(s)

error: Could not compile `i2cdev`.
```

Some upstream creates change their API ?

nastevens, 2017-06-23:
It looks like you're maybe trying to compile on OSX? This crate calls the Linux-only i2cdev interface.

I know there was some work on the `sysfs-gpio` to compile cleanly (but be useless) on OSX. It might be good to do something for this crate as well.

MrSpock, 2017-06-23:
Yes you are right. I'm working on macOS. 
Interesting - when I clone this repo locally and modify Cargo.toml:
```
[dependencies]
i2cdev = { path = "../rust-i2cdev-orig"}
```
then compilation is ok :):

```
spock@macspock:~/OneDrive/devel/vl53l0x> ck
    Finished dev [unoptimized + debuginfo] target(s) in 0.0 secs
spock@macspock:~/OneDrive/devel/vl53l0x>
```

nastevens, 2017-06-23:
Is your `ck` alias doing `cargo check`? Because that doesn't perform a full linking of the current crate, and since the error you were seeing was in the FFI portion it might not show up until the final linking step.

Does it still compile successfully if you do `cargo build` instead?

MrSpock, 2017-06-23:
Yes 'ck' is alias cor cargo check.
cargo build is also ok.
I will move with development to some Linux VM sooner or later.

martindeegan, 2017-08-03:
It looks like @fitzgen added OSX support 9 months ago, but the latest published crate is from over a year ago. Thats why it builds when you clone this repo.

nastevens, 2017-08-05:
@martindeegan that is definitely a problem - @posborne can we get a release pushed?

tadeegan, 2017-08-29:
Ping.

PhilipTrauner, 2018-01-26:
Ping.

posborne, 2018-01-26:
@tadeegan @PhilipTrauner Sorry for the long delay; I took a quick pass to prep for a new release and update some now ancient dependencies: https://github.com/rust-embedded/rust-i2cdev/pull/35.  Once merged will released to crates.io.

posborne, 2018-01-29:
0.3.2 is now released: https://crates.io/crates/i2cdev

closed 2018-01-29 by posborne