rust-embedded/rust-i2cdev #37: i2cdev recompiles every time

querry43, 2018-03-04:
i2cdev compiles every time I `cargo build`.

```
[dependencies]
i2cdev = "0.3.2"
```
`rustc 1.24.0 (4d90ac38c 2018-02-12)`

nastevens, 2018-03-04:
Looks like this is a [known issue of the `skeptic` crate](https://github.com/budziq/rust-skeptic/issues/25) that was [fixed after the 0.13.2 release](https://github.com/budziq/rust-skeptic/pull/68). I confirmed that if I add the following to my project's `Cargo.toml` the issue goes away:

```toml
[replace]
"skeptic:0.13.2" = { git = 'https://github.com/budziq/rust-skeptic' }
```

@querry43 could you please also verify that this addresses the issue in your case?

Obviously the long-term solution is for a new release of `skeptic` with the fix - I have opened an issue requesting the author to push a new release: https://github.com/budziq/rust-skeptic/issues/75

querry43, 2018-03-04:
Oh yea, that works great.  Thanks for finding that.  It fixes a problem I have with another crate too.

closed 2018-03-04 by querry43

nastevens, 2018-05-14:
Upstream release of rust-skeptic is done: https://github.com/budziq/rust-skeptic/issues/75

Since rust-i2cdev is used as a library, if you have an application with a `Cargo.lock` you can do `cargo update -p skeptic` to get the bumped version. I tested it out locally and i2cdev is no longer rebuilding every time `cargo build` is run.

querry43, 2018-05-14:
> Upstream release of rust-skeptic is done: budziq/rust-skeptic#75

Yea, I saw the upstream commit.  Thanks.