rust-embedded/rust-i2cdev #75: Seeking support in writing an I2C LCD library in Rust.

monomycelium, 2023-03-02:
### introduction
I have a Raspberry Pi and an 16-by-2 I2C LCD lying around. Last year, I found a Python3 library to control the LCD using SMBus. I want to write a similar library for Rust, but as someone who hardly knows how I2C works, I need some help.

### goal
Write a Rust library to simplify manipulating an I2C LCD (maybe by rewriting [`lcd_1602_i2c`](https://crates.io/crates/lcd_1602_i2c) for `i2cdev` for Linux).

### resources
- [implementation in Python3 by `vay3t`](https://gist.github.com/vay3t/8b0577acfdb27a78101ed16dd78ecba1)
- [`i2cdev` crate documentation](https://docs.rs/i2cdev/latest/i2cdev/linux/struct.LinuxI2CDevice.html#trait-implementations)
- [implementation in embedded Rust by John Socha-Leialoha](https://docs.rs/lcd_1602_i2c/latest/lcd_1602_i2c/)
- [implementation in JavaScript](https://youtu.be/98LPUqTTfjo)

So far, I have managed to get the backlight to turn on and off using the `i2cdev` crate (`src/main.rs`):
```
use i2cdev::core::*;
use i2cdev::linux::{LinuxI2CDevice, LinuxI2CMessage};
use std::time::Duration;
use std::thread;

const SLAVE_ADDR: u16 = 0x3F;

fn main(){
    let mut dev: LinuxI2CDevice = LinuxI2CDevice::new("/dev/i2c-1", SLAVE_ADDR).unwrap();
    let mut msgs: [LinuxI2CMessage; 1];
    
    loop {   
        msgs = [LinuxI2CMessage::write(&[0x08])];
        dev.transfer(&mut msgs).unwrap();

        thread::sleep(Duration::from_secs(1));
        
        msgs = [LinuxI2CMessage::write(&[0x00])];
        dev.transfer(&mut msgs).unwrap();

        thread::sleep(Duration::from_secs(1));
    }
}
```

### help needed
If you'd like to work with me in writing this library, please reach out to me. Or, you could help with what I am stuck on right now: sending "commands" to the LCD. I can write I2C bytes to the LCD just fine, but apparently, "commands" are different.

### edit
- share embedded Rust–implementation
- share JavaScript-implementation

monomycelium, 2023-03-02:
Being stupid, it had never occurred to me that I could search GitHub for such a library. Apparently, there are billion others, like [Patryk27/pwr-hd44780](/Patryk27/pwr-hd44780). Sorry for wasting your time if you are reading this.

closed 2023-03-02 by monomycelium

eldruin, 2023-03-02:
You may find the [awesome-embedded-rust list](https://github.com/rust-embedded/awesome-embedded-rust) useful to search for drivers, board support crates and much more.

monomycelium, 2023-03-05:
Thanks for that! However, I gave up on creating the library because there are many existing implementations out there.